KevinGong2013/Printer #1: Request

hemangshah, 2017-05-30:
Hi Kevin, I have a small problem here, this is my [class](https://github.com/hemangshah/printer), it has also the same name as yours: Printer. The purpose of my class is to print logs. It's a fancy replacement of default "print" function. Long story short: This is the first time, I am trying to add my class as a Pod on CocoaPods so that it's easily available for anyone who's using CocoaPods in their project. Now the thing is that I can't claim the same name as your as you have already taken the Printer for this repo. No problem, as I was not aware of the this already, I changed my spec name to "printer-logger" and changed the file name too [this is required]. All good well and I were able to add my class on CocoaPods. So now the time to test it so when I test I found two problems.

1. I can't import "Printer" as it's there but I can't. I needed to import "printer_logger" [- converted to _].
2. As I can't import the Printer class, I can't use any functions of it. Everything becomes none of the use.

In this case? What should I do? As I am really new to this, I am not sure which steps I can take further?

I am only thinking if no solution works if you can change your pod name to something else to justify my class? Anything you do or advice will be much appreciated.

KevinGong2013, 2017-06-01:
Perhaps adding the prefix for your pod is the best solution.

hemangshah, 2017-06-01:
Ok, thank you.

closed 2017-06-01 by hemangshah