KevinGong2013/Printer #4: Line Spacing is too narrow.

sigmo2nd, 2018-07-11:
Hey,

I had been looking for legit swift thermal printer library and yours is simply amazing.

It's faster and organized I just loved it. I really appreciated that you shared this.

I successfully imported to my project yesterday, however, I found something different with the previous one.

The space between two lines are really narrow, I tried not to bother you and figure it out myself.

Since I could make doubleHeight function alone, so I thought I could also do it this function as well.

However, I'm not sure how I can set default line spacing or send gap between lines?

Could you give me little tips, I would be very appreciated.

Also, I'm a noobie and I don't know about the policy or accustom on Github anyhow I missed it, please kindly guide me. Thanks millions.

KevinGong2013, 2018-07-11:
```"anything just for debug".bc.blank()``` will send gap between lines.

```PredefinedAttribute``` can set text style.

sigmo2nd, 2018-07-13:
Thanks! I figured it out by changing code here. The key was, Data.print(n) The default number was 70 but that was way too big for me. I think it would be great if you give some guide line to the people who use this library.

I think you library is the best one on the internet!

extension Receipt: Printable {

    public var datas: [Data] {

        var ds = blocks.map { Data.reset + $0.data + Data.print(10) }

        let data = Data(bytes: ESC_POSCommand.printAndFeed(lines: 7).rawValue)
        ds.append(data)

        return ds
    }
}

KevinGong2013, 2018-07-18:
ok, i will do that when have some time.

ntnhon, 2019-10-23:
Hi Kevin, thank you for the great library! I was facing the same issue, 70 is somewhat to big so I ended up modifying the source code myself. Note that when modifying something in cocoapods, you have to clean and rebuild the project or changes won't be taken into account.

KevinGong2013, 2019-10-23:
I'M BACK!

KevinGong2013, 2019-10-23:
Printer 4.0.1 has been support adjust FeedPoints per line.

```
func application(_ application: UIApplication, didFinishLaunchingWithOptions launchOptions: [UIApplication.LaunchOptionsKey: Any]?) -> Bool {
        ...
        Block.defaultFeedPoints = 70
        return true
    }
```

closed 2019-10-23 by KevinGong2013

ntnhon, 2019-10-23:
> Printer 4.0.1 has been support adjust FeedPoints per line.
> 
> ```
> func application(_ application: UIApplication, didFinishLaunchingWithOptions launchOptions: [UIApplication.LaunchOptionsKey: Any]?) -> Bool {
>         ...
>         Block.defaultFeedPoints = 70
>         return true
>     }
> ```

You're right I didn't think about changing the default value directly from my project, anyway it would be nice if you could make a default value at around 10 and 20.