KevinGong2013/Printer #6: Paper Size Height is Limited

guevarraluigi22, 2018-10-02:
Hello Guys Please help me on how to set the paper size to auto.. everything works well but when you print a receipt that has many details, let say 16 inches in height.. the printer only prints 10 inches of that then it will stop.. please help me guys :-(

moderateepheezy, 2019-06-19:
@guevarraluigi22 Where you able to solve this? If yes, please can you share your solution.

shahzebkhowaja, 2020-02-08:
@guevarraluigi22  If I understood your question correctly, you are trying to print content more than the limit of printer buffer, which could cause this issue to not print the complete content. 

While debugging this library, I found that there is `CharacteristicWriteType`  you need to set (ref: https://developer.apple.com/documentation/corebluetooth/cbcharacteristicwritetype). There are two types you can set, and currently in this library it's set to `.withoutResponse`, you should trying changing that to `.withResponse`. 

The difference between those two types is, one respond back to peripheral, and other doesn't. `.withResponse` allows the peripheral to receive the response back (if the data is write successfully or not)  and react on that. I hope this helps others.

alizainprasla, 2020-02-08:
@shahzebkhowaja work for me.

closed 2023-02-03 by KevinGong2013