KevinGong2013/Printer #8: Is there any library to use Thermal Printer to Print Over WIFI ??

jvashisht, 2019-03-11:
Thank you so much for this library, it helped me alot :)
But i also need to connect printer over wifi and print , Please help me in this if you have idea about this.
Thanks.

KevinGong2013, 2019-03-13:
Most bluetooth printer is kind of Thermal Printer.
Most modern printers support ESC/POS command set.
Send `Receipt Data`(ESC/POS command Data) over wifi, bluetooth or other transport layer very similar.

ramniwaseasyeat, 2020-10-23:
Yes, I have changed same code for wifi print, let me know if any one required this

xuehaoze, 2020-11-02:
> Yes, I have changed same code for wifi print, let me know if any one required this

Hello, would you kindly share some thought regarding to wifi printing? I've tried to connect with my wifi thermal printer for weeks, but never succeed. It would be great if you can give me some hints, thanks!

ramniwaseasyeat, 2020-11-03:
Sure i will share this one, it’s working in my ios and android devices. You
just need to replace code.

ncuthbert, 2020-12-06:
Hi @ramniwaseasyeat are you able to share your code for network printing? It would be a big help. I'd be happy to clean up any code to be included in this project. Thanks :)

ramniwaseasyeat, 2021-03-04:
1. Make TCP Client socket using any socket library and
2. Pass the printer IP address along with the port
3. Make sure your printer connect with the same wifi with your device
4. Convert Tickets data into UTF-8 encoded data

Exm -: 

let client = TCPClient(address: ipAddress, port: Int32(portNumber) ?? 9100)
          switch client.connect(timeout: 3) {
          case .success:
            self.printTicketDesignText(printData!, client, message,true)
            
          case .failure( _):
 ////
          }

areeb111, 2021-05-22:
@ramniwaseasyeat 
Please can you share the repo fork or send a PR to this repo to support WiFi/Ethernet printers.

closed 2023-02-03 by KevinGong2013

SHEBINKUNIYIL, 2023-05-16:
> Yes, I have changed same code for wifi print, let me know if any one required this

Could you please code developer.shebin@gmail.com