KevinGong2013/Printer #10: Print UIImage

phuongtd57, 2019-04-16:
Hi Kevin,

How to print image, pdf ???

KevinGong2013, 2019-04-17:
work in progress.

KevinGong2013, 2019-04-17:
I have added image cmd on master but my printer don't support  this cmd. can you help me debug these cmd? just clone and run example project.

phuongtd57, 2019-04-17:
Ok, i will try it tomorrow, i am at home now :)

phuongtd57, 2019-04-18:
Hi Kevin,
Image cmd does not work, it prints nothing

KevinGong2013, 2019-04-18:
😕

KevinGong2013, 2019-04-18:
print nothing means printer will feed paper ?

phuongtd57, 2019-04-18:
Bank payper

phuongtd57, 2019-04-18:
i send print result to you in email

phuongtd57, 2019-04-18:
blank paper

huynhphan89, 2019-04-27:
Did you guys solve this issue ?

pylearndl, 2019-05-18:
> blank paper

> Did you guys solve this issue ?

Hi,

I have forked Kevin's git and update new ImageBlock which prints esc/pos image on thermal receipt printer. Check it our on my forked link

https://github.com/pylearndl/Printer

closed 2019-06-20 by KevinGong2013