KevinGong2013/Printer #11: Type 'Block' has no member 'dividing'

RobertoPanzera, 2019-04-18:
Hi I wanted to use your library but it doesn't work,
I took the code from the example and I get this error:
 var receipt = Receipt(
                .dividing,  **Type 'Block' has no member 'dividing'**
                .qr("Icey.Liao"),
                .dividing    )

KevinGong2013, 2019-04-18:
`Printer` framework not build ?

RobertoPanzera, 2019-04-19:
I added this line to the pod file:
pod 'Printer', '~> 2.0'

KevinGong2013, 2019-04-19:
pod 'Printer',  :branch => 'master'

RobertoPanzera, 2019-04-24:
with this > pod 'Printer', :branch => 'master'
i have this error: [!] Unknown external source parameters for `Printer`: `{:branch=>"master"}`

if I write this in pod file:
pod 'Printer', :git => 'https://github.com/KevinGong2013/Printer.git', :branch => 'master'
pod install work and have this message Installing Printer 2.0.5 (was 2.0.3)

but I have error in xcode: "No such module 'Printer' " in Viewcontroller and "Use of unresolved identifier 'CentralManagerDelegate'; did you mean 'CBCentralManagerDelegate'?" in Printer folder

KevinGong2013, 2019-06-20:
new version released.

closed 2019-06-20 by KevinGong2013