KevinGong2013/Printer #12: Compatible Printer?

QiquanLu, 2019-05-13:
Hi, Kevin
Could you recommend a few printers that compatible with your code?
I thought anything claimed support ESC/POS would be okay, but I tried this one, https://www.amazon.com/Bluetooth-Portable-Wireless-Rechargeable-Compatible/dp/B0757JG57P ,it's not working.

Thanks

boardmain, 2019-06-04:
yes
can you suggest some printer?

boardmain, 2019-06-11:
i bought this [Issyzone POS](https://www.aliexpress.com/item/IMP006-Hot-sales-pos58-thermal-printer-Mini-Bluetooth-Android-thermal-printer-receipt-printer-Reliability-100km-Print/32704704405.html?aff_platform=aaf&afref=&dp=447d59ff4ffc3328735ce4de7a2ac170&onelink_item_from=32704704405&algo_expid=b8c5bd6a-8136-48d5-94be-bdbf52da4937-15&onelink_thrd=0.0&ws_ab_test=searchweb0_0%252Csearchweb201602_10_10065_10068_319_317_10696_10084_453_10083_454_10618_10304_10307_10820_10821_537_10302_536_10843_10059_10884_10887_321_322_10103%252Csearchweb201603_52%252CppcSwitch_0&pvid=b68aa920-a3f1-4c77-9313-1c1c18d9f9ea&onelink_duration=0.748916&sk=VnYZvQVf&onelink_status=noneresult&scm=1007.22893.125764.0&terminal_id=c6eb84bc05704d3e84711db37c9fb4c7&algo_pvid=b8c5bd6a-8136-48d5-94be-bdbf52da4937&af=1011933&cpt=1560242941640&spm=2114.search0104.3.87.480c1b27db8Bak&onelink_page_from=ITEM_DETAIL&cv=47843&onelink_item_to=32704704405&mall_affr=pr3&aff_trace_key=1afa2bd7962b4220bd9e7c766fc95f1a-1560242941640-03390-VnYZvQVf&onelink_page_to=ITEM_DETAIL)  it work well with this library

justinshuls, 2019-07-25:
Ever get your PUQU working? I am also trying to use a PUQU Q00

Qiquan-K-Lu, 2019-07-25:
@justinshuls I couldn't get PUQU Q20 to work with this library, so I assume same thing with their Q00.
I ended up with [MUNBYN](https://www.amazon.com/gp/product/B07N86R5RB) brand. It's seems less rigid than the PUQU Q20, but it works great with this library.

justinshuls, 2019-07-28:
What's weird is that the image block works but everything else doesn't. I parsed through the code to see if anything is different about the image block but unable to conclude anything. Love my little sticker printer, but really need to tie it into my systems

justinshuls, 2019-08-18:
@qiquan-K-Lu does that printer support labels?

sawyermade, 2024-12-24:
So I havent tried this package out yet but esc/pos is pretty standard across platforms. I have a munbyn imp002 and have it implemented with labels. You just have to make sure to get smaller label rolls, usually around 100-150 labels per roll or it wont fit in it since it's small and portable. Those big 100 label rolls wont work with it. I am currently using 50x80mm rolls with I think 115 labels per roll. 

Im going to test this out this week on iOS with my munbyn imp002 but my guess is that it will work since I was able to pair it with my iPhone no problem and it works great on android with a similar esc/pos library. Here are the links for the labels and printer I have.

Printer Munbyn IMP002: https://a.co/d/3vCrnNc

Labels 50x80mm: https://a.co/d/3vCrnNc

KevinGong2013, 2026-10-17:
Compatibility is tracked in a single pinned issue from now on; 5.0.0 scans for the common BLE printer services (Gprinter/Xprinter `E7810A71…`, ISSC `49535343…`, `18F0`, `FFE0`, `FF00`) and picks the write characteristic by its properties, so most ESC/POS printers with a BLE module work out of the box. Printers that only offer classic Bluetooth (SPP) cannot be used from iOS. Please add your model and result to the compatibility issue.

closed 2026-10-17 by KevinGong2013