KevinGong2013/Printer #15: image printing error and QR not printing

DhanjuAjay, 2019-07-06:
guard let image = UIImage(named: "demo") else {
                return
            }
            
            var receipt = Receipt(
                .dividing,
                .qr("Icey.Liao"),
                .dividing,
                .image(image, attributes: [ImageBlock.PredefinedAttribute.alignment(.center)])
                //  Updated by Pradeep Sakharelia on 15/05/19
            )
            
            receipt.feedLinesOnTail = 2
            receipt.feedPointsPerLine = 60
            
            pm.print(receipt)

[image]

I am using above code to print image and Qr code but it shows error while printing, As Shown in image

pylearndl, 2019-07-18:
Upload your demo image here. Is it jpg image you want to print? What is image size? Hight x width of image ?

DhanjuAjay, 2019-07-18:
It is the same image that is in example of project with name demo.png

abu4tw, 2019-07-18:
Image is work to me , but .qr only print string not QRcode, any one know how to make image printing alignment in center or print rectangle image

pylearndl, 2019-07-18:
Try converting qr to image, then print

abu4tw, 2019-07-19:
> Try converting qr to image, then print

thanks. i will try..

DhanjuAjay, 2019-07-20:
> let img = UIImage(named: "demo")! var receipt = Receipt() receipt.feedLinesOnTail = 1 let imgblock = ImageBlock(img, attributes: [ImageBlock.PredefinedAttribute.alignment(.center)]) receipt.add(block: imgblock) for datas in receipt.datas { let sendtest = self.client.send(data: datas as NSData) print("send data: \(sendtest)") }

It shows errors on these lines >
1. receipt.add(block: imgblock)  >>>>  receipt.add(block: imgblock)
2. for datas in receipt.data  >>>>  Type '(String.Encoding) -> [Data]' does not conform to protocol 'Sequence'

pylearndl, 2019-07-20:
> > let img = UIImage(named: "demo")! var receipt = Receipt() receipt.feedLinesOnTail = 1 let imgblock = ImageBlock(img, attributes: [ImageBlock.PredefinedAttribute.alignment(.center)]) receipt.add(block: imgblock) for datas in receipt.datas { let sendtest = self.client.send(data: datas as NSData) print("send data: (sendtest)") }
> 
> It shows errors on these lines >
> 
> 1. receipt.add(block: imgblock)  >>>>  Cannot convert value of type 'ImageBlock' to expected argument type 'Block'
> 2. for datas in receipt.data  >>>>  Type '(String.Encoding) -> [Data]' does not conform to protocol 'Sequence'

try
let img = UIImage(named: "demo")!
            var receipt = Receipt()
            receipt.feedLinesOnTail = 2
            receipt.add(block: Block.image(img, attributes [ImageBlock.PredefinedAttribute.alignment(.center)]))
            pm.print(receipt)

DhanjuAjay, 2019-07-21:
> try
> let img = UIImage(named: "demo")!
> var receipt = Receipt()
> receipt.feedLinesOnTail = 2
> receipt.add(block: Block.image(img, attributes [ImageBlock.PredefinedAttribute.alignment(.center)])) [...]

Unfortunately same result, it just print 20% of image and then blank
[image]

Mahipal-codzgarage, 2019-08-19:
@DhanjuAjay can you solved printing image issue?

DhanjuAjay, 2019-08-31:
> @DhanjuAjay can you solved printing image issue?

Still not solve, even the example app is just printing title, then stop working.

pylearndl, 2019-08-31:
@DhanjuAjay what thermal printer are you using? Have you tried another thermal receipt printer. It works in ESC/POS printers

DhanjuAjay, 2019-09-21:
> @DhanjuAjay which thermal printer are you using? Have you tried another thermal receipt printer. It works on ESC/POS printers
> 
> Try 80mm thermal receipt printer from eBay will do the job

I am using this printer:

https://www.amazon.com.au/NYEAR58-10mm-Personal-Wireless-Bluetooth-Compatible/dp/B07C8RDLZZ/ref=asc_df_B07C8RDLZZ/?tag=googleshopdsk-22&linkCode=df0&hvadid=341773403893&hvpos=1o1&hvnetw=g&hvrand=9311021397934973382&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9071510&hvtargid=pla-625816585876&psc=1

Not working for me

pylearndl, 2019-09-21:
https://www.ebay.co.uk/c/1874761593
2017 WiFi POS Wireless Thermal Receipt Printer 80mm Auto ...

Successfully tested and working on these printers

DhanjuAjay, 2019-09-21:
Thanks for information, is there any way to send printing commands from my app, Thanks

pylearndl, 2019-09-21:
I think your printer is for label printing. I haven't tested my code on label printer.

I'm sure image will print, if you try print on 80mm receipt printer, is mainly used in epos receipt printing.

DhanjuAjay, 2019-09-21:
> I think your printer is for label printing. I haven't tested my code on label printer.
> 
> I'm sure image will print, if you try print on 80mm receipt printer, is mainly used in epos receipt printing.

I have printer this print from android app, and printer works perfectly with an android library, but not with iOS library 😔
[image]

DhanjuAjay, 2019-09-27:
Hello , I have tested the latest example, now alignment is not working, initially example was working fine for text only 
[image]

ntnhon, 2019-10-23:
I am encountering the same issue, it just prints blank spaces. It even crashes sometimes. Is there a specific dimension for the images?

pylearndl, 2019-10-23:
Print only square logo image. Try one of 32x32 px, 64x64, 128x128, 256x256. If your image is not any of these size than resize programmatically before give it to printer's receipt object.

ntnhon, 2019-10-23:
@pylearndl I download your fork and run the sample project, the image is 40x40 so I resized it to 32x32, still it doesn't work. It will just print blank spaces. Did you test your fork on a real device?

pylearndl, 2019-10-23:
Send me screenshot and error, what printer you using?
My all deployed ios apps are working on wifi 80mm thermal printer just fine. Kevins' code wasn't working initially when I tried to integrate. I made my code to make it work and fork in his git to let others to use it.

ntnhon, 2019-10-23:
@pylearndl thank for the quick reply. [Here is my printer specs](https://my.pcloud.com/publink/show?code=XZlhx07ZU2iPgT9UjchhPyacg9D7SQRovyP7)

No error is logged to the console. The printer slightly prints a blank space. That's all my observation.

pylearndl, 2019-10-23:
What printer you using? Send me image

pylearndl, 2019-10-23:
I can see you trying to print on 58mm printer. Unfortunately I m not sure about it. I have tested and it's working absolutely fine on 80mm thermal receipt printers.

pylearndl, 2019-10-23:
https://www.ebay.co.uk/c/1874761593
2017 WiFi POS Wireless Thermal Receipt Printer 80mm Auto ...

pylearndl, 2019-10-23:
I sent you this link to see the printer ios code works. I haven't tested on any other thermal printers

ntnhon, 2019-10-23:
@pylearndl it might be the problem of printer model. As I'm making a receipt printing app for a small business, a 80mm printer is way too big. I will wait for upcoming updates, for now I'm happy printing the logo as a large text. Thank you for your time 👍

pylearndl, 2019-10-23:
Kevin has updated his code on GitHub with new version. You should try and see if it works for your 58mm printer. Just give it a go

chonnawee4, 2020-07-08:
I'm also can't print image on 58mm printer

KevinGong2013, 2023-02-03:
`Printer` has been rewrite.

select `Mode` to `double width& double height`

```
let receipt = Receipt(.🖨️58(.ascii))
<<~ .style(.initialize)
<<< Image(cgImage, mode: .doubleWH)
<<~ .page(.printAndFeed(lines: 1))
```

**Image(cgImage, mode: .doubleWH)** is key code.

kruyvanna, 2023-08-02:
@pylearndl how to print non-square image?

KevinGong2013, 2026-10-17:
Fixed in 5.0.0 — the image path was rewritten: correct gray conversion (the old formula capped gray at 106, so the default threshold produced a blank image), any `CGImage` format (no more "Index out of range"), auto-scale to the paper width, banded `GS v 0` output, Floyd–Steinberg dithering, and Bluetooth writes are now split into 20-byte packets (large packets are silently truncated by many printer modules — verified on a Gprinter GP-58). QR codes use the correct command sequence as well.

Please update to 5.0.0 (Swift Package Manager; CocoaPods was dropped) and reopen if it still fails — include the base64 from `DummyPrinter().history` and your printer model.

closed 2026-10-17 by KevinGong2013