KevinGong2013/Printer #23: When convert Image to Data

mchan004, 2019-11-16:
I try to get Data from Ticket
When convert Image to Data, it will Fatal error: Index out of range
How to resolve it?
[image]

rikbauwens, 2019-12-14:
It's a bug. But if you make sure the width of your image is dividable by 8 (e.g. 160), it'll work.

KevinGong2013, 2023-02-03:
`Printer` framework has been rewrite.

can you help testing qrcode&image command?

```
let receipt = Receipt(.🖨️58(.ascii))
<<~ .style(.initialize)
<<< QRCode(content: "https://g.cn")
<<< Image(cgImage)
```

KevinGong2013, 2026-10-17:
Fixed in 5.0.0 — the image path was rewritten: correct gray conversion (the old formula capped gray at 106, so the default threshold produced a blank image), any `CGImage` format (no more "Index out of range"), auto-scale to the paper width, banded `GS v 0` output, Floyd–Steinberg dithering, and Bluetooth writes are now split into 20-byte packets (large packets are silently truncated by many printer modules — verified on a Gprinter GP-58). QR codes use the correct command sequence as well.

Please update to 5.0.0 (Swift Package Manager; CocoaPods was dropped) and reopen if it still fails — include the base64 from `DummyPrinter().history` and your printer model.

closed 2026-10-17 by KevinGong2013