KevinGong2013/Printer #29: Printer are not listing on controller

alizainprasla, 2020-01-05:
Swift version: 4.2
Pod version: 4.0.1
Printer: [Meihengtong Label Printer, Thermal 80MM Receipt Printer](https://www.amazon.com/Printers-Bluetooth-Rechargeable-Business-Supermarket/dp/B07HCV6T2G)

By using the current example project. i've tried to detect printers. **BluetoothPrinterSelectTableViewController** is not showing any printer around.

shahzebkhowaja, 2020-01-06:
I also have the same issue. Tweak some code to allow all type of Bluetooth devices, but still not able to see the bluetooth printer or any other bluetooth device.

closed 2020-02-08 by alizainprasla

alizainprasla, 2020-02-08:
Closing this issue as I found the reason for not showing my printer device in sample App.

https://stackoverflow.com/a/51122125/612854

> Short story: A bluetooth thermal printer can work with iOS only if it uses the Bluetooth 4.0 LE version or if the manufacturer registered in the Mfi Program.
> 
> Bluetooth LE 4.0 devices aren't listed in the Settings => Bluetooth on iOS anyway.
> 
> If your bluetooth thermal printer has Bluetooth 4.0 LE you can adapt the official Apple sample code using Objective C sample code or in Swift sample code
> 
> Long story: iOS and macOS SDKs supports Bluetooth 4.0 LE devices with the Core Bluetooth framework and other Bluetooth versions devices with the External Accessory framework. The External Accessory framework requires that the manufacturer of the bluetooth thermal printer is registered in the Mfi Program. Only the major manufacturers like Epson, Star Micronics, Zebra, Bixolon are registered with Mfi Program.
> 
> If you buy a cheap bluetooth thermal printer online from a minor manufacturer it will only work with iOS if it uses Bluetooth 4.0 LE since it uses the Core Bluetooth framework which does not require any registration to the Mfi Program.