KevinGong2013/Printer #30: Problems printing German letters

chnbz, 2020-02-03:
Hi,

the printer doesn't print German letters like Ä,Ö or Ü instead it prints "ü0ë5" for example. Tried different code pages that the printer supports but not much has changed. Any idea how to resolve this?

chnbz, 2020-02-03:
Solved it. Found out the print() function has an optional parameter called encoding 
[image]

I changed the code page to WPC1252 as well and since then everything works fine.

mali1488, 2020-06-14:
How did you set the codePage? 
Having problems with åäö.

chnbz, 2020-07-02:
> How did you set the codePage?
> Having problems with åäö.

With a Windows app that the printer manufacturer provided :)

closed 2023-02-03 by KevinGong2013