KevinGong2013/Printer #31: Fatal error: Index out of range Image printing

KansV3Cube, 2020-02-22:
While printing image getting this error and application crashed.

Please help me out

kaitoshadow, 2022-01-25:
I got same issue :( Have you found solution?

KevinGong2013, 2023-02-03:
`Printer` framework has been rewrite.

can you help testing qrcode&image command?

```
let receipt = Receipt(.🖨️58(.ascii))
<<~ .style(.initialize)
<<< QRCode(content: "https://g.cn")
<<< Image(cgImage)
```

manojssas, 2023-02-05:
Hi is there any way to Print Arabic langauge?

manojssas, 2023-02-05:
@KevinGong2013 How to print arabic Langauge if coming from server?

KevinGong2013, 2023-02-06:
```

public extension String {
    
    struct ArabicEncoding {
        public static let windowsArabic = String.Encoding(rawValue: CFStringConvertEncodingToNSStringEncoding(CFStringEncoding(CFStringEncodings.windowsArabic.rawValue)))
    }
}

let receipt = Receipt(.🖨️58(String.ArabicEncoding.windowsArabic))

```

KevinGong2013, 2026-10-17:
Fixed in 5.0.0 — the image path was rewritten: correct gray conversion (the old formula capped gray at 106, so the default threshold produced a blank image), any `CGImage` format (no more "Index out of range"), auto-scale to the paper width, banded `GS v 0` output, Floyd–Steinberg dithering, and Bluetooth writes are now split into 20-byte packets (large packets are silently truncated by many printer modules — verified on a Gprinter GP-58). QR codes use the correct command sequence as well.

Please update to 5.0.0 (Swift Package Manager; CocoaPods was dropped) and reopen if it still fails — include the base64 from `DummyPrinter().history` and your printer model.

closed 2026-10-17 by KevinGong2013