KevinGong2013/Printer #33: Compatible printer

siki88, 2020-03-17:
please help. how to implementation 
[http://www.gprinter.hk/download/2/#sdk](http://www.gprinter.hk/download/2/#sdk)
to yours Printer framework ?
thx

KevinGong2013, 2020-03-18:
Wow, this really useful. After holiday, I will refactor this project according the official documents.

KevinGong2013, 2026-10-17:
Compatibility is tracked in a single pinned issue from now on; 5.0.0 scans for the common BLE printer services (Gprinter/Xprinter `E7810A71…`, ISSC `49535343…`, `18F0`, `FFE0`, `FF00`) and picks the write characteristic by its properties, so most ESC/POS printers with a BLE module work out of the box. Printers that only offer classic Bluetooth (SPP) cannot be used from iOS. Please add your model and result to the compatibility issue.

closed 2026-10-17 by KevinGong2013