KevinGong2013/Printer #35: Index out of range when printing image

MrGD0924, 2020-05-04:
how to fix it? Please give me response....

MrGD0924, 2020-05-04:
import Printer

class ViewController: UIViewController {

    let pm = PrinterManager()
    let a = "3025856504282187562209314690839033975462381239936941770344840034144563413477955812966608888478861688170367239204693672108967226008937498861858661874577081699385849994003420752137009590997581532249708930640325165906723111760564674259827980794930442717664614029549142170395528352178494728525469158037062010011689105259"

    override func viewDidLoad() {
        super.viewDidLoad()
    }

    @IBAction func touchPrint(sender: UIButton) {

        let image = generateQRCode(from: a)
        if pm.canPrint {

            var receipt = Receipt(
                .dividing,
                .image(image!, attributes: [ImageBlock.PredefinedAttribute.alignment(.center)]),
                //  Updated by Pradeep Sakharelia on 15/05/19
                .dividing
            )
            
            receipt.feedLinesOnTail = 2
            receipt.feedPointsPerLine = 60
            
            pm.print(receipt)
        } else {

            performSegue(withIdentifier: "ShowSelectPrintVC", sender: nil)
        }
    }

    override func prepare(for segue: UIStoryboardSegue, sender: Any?) {
        if let vc = segue.destination as? PrinterTableViewController {
            vc.sectionTitle = "Choose Printer"
            vc.printerManager = pm
        }
    }
    
    func generateQRCode(from string: String) -> UIImage? {
       // Get data from the string
        let data = string.data(using: String.Encoding.ascii)
        // Get a QR CIFilter
        guard let qrFilter = CIFilter(name: "CIQRCodeGenerator") else { return nil}
        // Input the data
        qrFilter.setValue(data, forKey: "inputMessage")
        // Get the output image
        guard let qrImage = qrFilter.outputImage else { return nil}
        // Scale the image
        let transform = CGAffineTransform(scaleX: 10, y: 10)
        let scaledQrImage = qrImage.transformed(by: transform)
        // Invert the colors
        guard let colorInvertFilter = CIFilter(name: "CIColorInvert") else { return nil}
        colorInvertFilter.setValue(scaledQrImage, forKey: "inputImage")
        guard let outputInvertedImage = colorInvertFilter.outputImage else { return nil}
        // Replace the black with transparency
        guard let maskToAlphaFilter = CIFilter(name: "CIMaskToAlpha") else { return nil}
        maskToAlphaFilter.setValue(outputInvertedImage, forKey: "inputImage")
        guard let outputCIImage = maskToAlphaFilter.outputImage else { return nil}
        // Do some processing to get the UIImage
        let context = CIContext()
        guard let cgImage = context.createCGImage(outputCIImage, from: outputCIImage.extent) else { return nil}
        return UIImage(cgImage: cgImage)
    }
}

AlekseyPakAA, 2020-07-08:
Try to create an image where width % 4 == 0. I think there's a bug related to rounding.

ramniwaseasyeat, 2020-12-28:
Any solution on this, I'm also facing same problem

KevinGong2013, 2023-02-03:
Image command has rewrite.
please test with latest code.

KevinGong2013, 2026-10-17:
Fixed in 5.0.0 — the image path was rewritten: correct gray conversion (the old formula capped gray at 106, so the default threshold produced a blank image), any `CGImage` format (no more "Index out of range"), auto-scale to the paper width, banded `GS v 0` output, Floyd–Steinberg dithering, and Bluetooth writes are now split into 20-byte packets (large packets are silently truncated by many printer modules — verified on a Gprinter GP-58). QR codes use the correct command sequence as well.

Please update to 5.0.0 (Swift Package Manager; CocoaPods was dropped) and reopen if it still fails — include the base64 from `DummyPrinter().history` and your printer model.

closed 2026-10-17 by KevinGong2013