KevinGong2013/Printer #39: The result print out is empty

sandipermata, 2020-07-05:
Hi KevinGong,

i have try created new project and implemented this library.
when i searching the printer, the printer found and success connected.
i have issue when i try to print out, the result on the paper is empty.

detail my printer 
my printer : Thermal receipt printer
model: PRJ CX-58B
Interface : USB+BT2.0+BT4.0
Brand : Panda

how to fix this issue ?

umairgillani94, 2020-09-09:
having same issue any solution found?

KevinGong2013, 2023-02-03:
```

let receipt = Receipt(.🖨️58(.ascii))
<<~ .style(.selectsCharacterFont(.c))
<<< "testing1"
<<< "testing2"
<<< "testing3"
<<< "testing4"
<<~ .page(.print)

```

Try this code.

KevinGong2013, 2026-10-17:
Fixed in 5.0.0 — the image path was rewritten: correct gray conversion (the old formula capped gray at 106, so the default threshold produced a blank image), any `CGImage` format (no more "Index out of range"), auto-scale to the paper width, banded `GS v 0` output, Floyd–Steinberg dithering, and Bluetooth writes are now split into 20-byte packets (large packets are silently truncated by many printer modules — verified on a Gprinter GP-58). QR codes use the correct command sequence as well.

Please update to 5.0.0 (Swift Package Manager; CocoaPods was dropped) and reopen if it still fails — include the base64 from `DummyPrinter().history` and your printer model.