KevinGong2013/Printer #41: Remove extra line feed/return?

michaelmruta, 2020-07-17:
there is a new line for every text i print, like the example below
how can i fix this?

.plainText("Text A"),
.plainText("Text")

```
Text A

Text B
```

.plainText("Text A\n Text B"), this works but i want to use the block w/ text alignment
```
Text A
Text B
```

abaif, 2020-08-09:
This is a really big deal, printing the sample ticket I got about 10 inches long! Also need to put text grouped for showing some tablet struct.

For now, I`m using processed images to fill these need, until got solutions this is a workaround.

KevinGong2013, 2023-02-03:
```
let receipt = Receipt(.🖨️58(.ascii))
<<~ .style(.selectsCharacterFont(.c))
 <<< "testing1"
<<< "testing2"
<<< "testing3"
<<< "testing4"
 <<~ .page(.print)
```

KevinGong2013, 2026-10-17:
5.0.0 adds `Text("…")`, which prints without a trailing line feed, so several items can share a line; a plain `String` still ends the line.

closed 2026-10-17 by KevinGong2013