KevinGong2013/Printer #47: multiple item

chitranjank, 2021-01-27:
how to print multiple item in one by one dynamically?

KevinGong2013, 2021-01-28:
sure.

chitranjank, 2021-01-28:
any example?

chitranjank, 2021-02-01:
> how to print multiple item in one by one dynamically?
> can I write (var ticket = Ticket()) for loop?

chitranjank, 2021-02-26:
hello Kevin

tanalavijay, 2021-09-21:
use add block concept for this 
example:
                        self.ticket.add(block: Block(Text(content: tipData, predefined: .bold , .alignment(.left))))
                        self.ticket.add(block: Block(Text(content: totData, predefined: .bold , .alignment(.left))))

KevinGong2013, 2023-02-03:
```

let receipt = Receipt(.🖨️58(.ascii))
        
receipt.append(item: "dynamically")
receipt.append(item: "content" <<~ .style(.underlineMode(.enable2dot)))
receipt.append(item: Image(cgImage) <<~ .layout(.justification(.center)))

```

closed 2023-02-03 by KevinGong2013