KevinGong2013/Printer #48: TicketImage not visible

mb812, 2021-07-12:
I'm not able to print a 128x128 full black image (or any other image).

TicketImage code:
`            
            Block.defaultFeedPoints = 30
            
            var ticket = Ticket()
            ticket.feedLinesOnTail = 3
            ticket.feedLinesOnHead = 0

            ticket.add(block: .image(UIImage(named: "black")!, attributes: TicketImage.PredefinedAttribute.alignment(.center)))
            ticket.add(block: .title("TEST"))
            ticket.add(block: .text(.init(content: "TEST", predefined: .alignment(.left))))
            ...

            ... print
`

Should this depend on the printer device?
I'm using a GOOJPRT PT-210 thermal printer.

Everything else works great!

developerGM, 2021-07-26:
@KevinGong2013 any news about this task?

burin-t, 2022-05-20:
Check the scale of image (image.scale) or resize it to 1x scale and image width must mod 8 = 0

KevinGong2013, 2023-02-03:
Printer has been rewrite.

Please try this code.

```
let receipt = Receipt(.🖨️58(.ascii))
<<~ .style(.initialize)
 <<< Image(cgImage)
<<~ .page(.printAndFeed(lines: 1))
```

KevinGong2013, 2026-10-17:
Fixed in 5.0.0 — the image path was rewritten: correct gray conversion (the old formula capped gray at 106, so the default threshold produced a blank image), any `CGImage` format (no more "Index out of range"), auto-scale to the paper width, banded `GS v 0` output, Floyd–Steinberg dithering, and Bluetooth writes are now split into 20-byte packets (large packets are silently truncated by many printer modules — verified on a Gprinter GP-58). QR codes use the correct command sequence as well.

Please update to 5.0.0 (Swift Package Manager; CocoaPods was dropped) and reopen if it still fails — include the base64 from `DummyPrinter().history` and your printer model.

closed 2026-10-17 by KevinGong2013