KevinGong2013/Printer #54: Printer is not listing on controller

smalldevcloud, 2022-04-16:
[image]
[image]
After adding the framework, I do not see a single device in the list of devices. Here's the reason - https://stackoverflow.com/questions/47056562/how-to-connect-thermal-bluetooth-printer-to-ios-devices/51122125#51122125

I found another framework on github. He, as seen in the screenshot, is able to find such 4.0 devices.
https://github.com/RickRedSix/BLE4.0-iOS-Swift-Demo

How to combine these two frameworks to get support for 4.0 BT printers?

smalldevcloud, 2022-04-18:
https://file.cdn.sunmi.com/SUNMIDOCS/CloudPrinter_IOSSDK_Demo.rar
Also, found this demo for iOS. this is from the manufacturer's website, it finds the printer and communicates with it via bluetooth, apparently, but there is no print example

KevinGong2013, 2023-02-03:
*How to combine these two frameworks to get support for 4.0 BT printers?*

Just use `Printer` to assemble receipt data.

Write data by another one.

closed 2023-02-03 by KevinGong2013