KevinGong2013/Printer #55: Wrong image on Ticket

frompon, 2022-04-22:
When I print an image on ticket I get this result. How can I solve this?
[image]

smalldevcloud, 2022-04-23:
With the image with smaller size you're gettin' the same result?

TillMoss, 2022-04-24:
This result is on Epson TM-T20 III if I use another brand works correctly

KevinGong2013, 2023-02-03:
`Printer` has been rewrite.

select `Mode` to `double width& double height`

```
let receipt = Receipt(.🖨️58(.ascii))
<<~ .style(.initialize)
<<< Image(cgImage, mode: .doubleWH)
<<~ .page(.printAndFeed(lines: 1))
```

**Image(cgImage, mode: .doubleWH)** is key code.

KevinGong2013, 2026-10-17:
Fixed in 5.0.0 — the image path was rewritten: correct gray conversion (the old formula capped gray at 106, so the default threshold produced a blank image), any `CGImage` format (no more "Index out of range"), auto-scale to the paper width, banded `GS v 0` output, Floyd–Steinberg dithering, and Bluetooth writes are now split into 20-byte packets (large packets are silently truncated by many printer modules — verified on a Gprinter GP-58). QR codes use the correct command sequence as well.

Please update to 5.0.0 (Swift Package Manager; CocoaPods was dropped) and reopen if it still fails — include the base64 from `DummyPrinter().history` and your printer model.

closed 2026-10-17 by KevinGong2013