KevinGong2013/Printer #57: not responding on print

ritesh305, 2022-07-26:
can anyone confirm if this [printer](https://amzn.eu/d/ih1SmIv) is compatible with this?  the printer is connected in the select printer section but there is no response in the printer when i click on print button from the example project given in this project,  any help would be appreciable. Thank you.

KevinGong2013, 2023-02-03:
Printer has been rewrite.

Can you testing again?

KevinGong2013, 2026-10-17:
Fixed in 5.0.0 — the image path was rewritten: correct gray conversion (the old formula capped gray at 106, so the default threshold produced a blank image), any `CGImage` format (no more "Index out of range"), auto-scale to the paper width, banded `GS v 0` output, Floyd–Steinberg dithering, and Bluetooth writes are now split into 20-byte packets (large packets are silently truncated by many printer modules — verified on a Gprinter GP-58). QR codes use the correct command sequence as well.

Please update to 5.0.0 (Swift Package Manager; CocoaPods was dropped) and reopen if it still fails — include the base64 from `DummyPrinter().history` and your printer model.

closed 2026-10-17 by KevinGong2013