KevinGong2013/Printer #58: Why image print as Chinese text?

karthikeyan1425, 2022-07-26:


KevinGong2013, 2023-02-03:
`Printer` has been rewrite.

Please try this code.

```
let receipt = Receipt(.🖨️58(.ascii))
<<~ .style(.initialize)
 <<< Image(cgImage)
<<~ .page(.printAndFeed(lines: 1))
```

KevinGong2013, 2026-10-17:
Fixed in 5.0.0. The `ESC t` code-page byte was wrong before, so selecting a code page never worked. See the "Encodings" table in the README: pair the receipt encoding with the printer code page, e.g. `Receipt(.🖨️58(.windowsCP1252))` + `.style(.selectCharacterCodePage(.wpc1252))` for ä ö ü €, or `String.ArabicEncoding.windowsArabic` + `.wpc1256` for Arabic. Characters the encoding cannot represent are now replaced instead of dropping the whole line. Cash drawer: `.peripheral(.cashDrawer())`.

closed 2026-10-17 by KevinGong2013