KevinGong2013/Printer #60: Not able to print Barcode Image

PraveenSharma001, 2022-08-06:
I am trying to print barcode, for that I am converted barcode into image and trying to print image but image is not printing, can anyone help?

KevinGong2013, 2023-02-03:
`Printer` framework has been rewrite.

can you help testing qrcode&image command?

```
let receipt = Receipt(.🖨️58(.ascii))
<<~ .style(.initialize)
<<< QRCode(content: "https://g.cn")
<<< Image(cgImage)
```

KevinGong2013, 2026-10-17:
Fixed in 5.0.0 — the image path was rewritten: correct gray conversion (the old formula capped gray at 106, so the default threshold produced a blank image), any `CGImage` format (no more "Index out of range"), auto-scale to the paper width, banded `GS v 0` output, Floyd–Steinberg dithering, and Bluetooth writes are now split into 20-byte packets (large packets are silently truncated by many printer modules — verified on a Gprinter GP-58). QR codes use the correct command sequence as well.

Please update to 5.0.0 (Swift Package Manager; CocoaPods was dropped) and reopen if it still fails — include the base64 from `DummyPrinter().history` and your printer model.

5.0.0 fixes the QR command sequence (the model / module-size / print function codes were wrong, so many printers printed nothing) and adds a `Barcode` item (CODE128, EAN-13/8, UPC, CODE39, ITF, CODABAR, CODE93). For non-ASCII QR content use `QRCode(content: "中文", encoding: .utf8)`. Closing; reopen if it still fails on your printer.

closed 2026-10-17 by KevinGong2013