KevinGong2013/Printer #61: connect to printer

aus60, 2022-09-03:
how can I connect to printer with this project 
thank you

KevinGong2013, 2023-02-03:
tap `Choose Printer` button & choose your printer.

linhgfx, 2023-06-12:
> tap `Choose Printer` button & choose your printer.

i wrote this code to open printer select screen.

func listDeviceBlueTooth(){
        let storyboard = UIStoryboard(name: "Main", bundle: nil);
        let vc = storyboard.instantiateViewController(withIdentifier: "ListBluetoothDevice") as! UIViewController; // MySecondSecreen the storyboard ID
        self.present(vc, animated: true, completion: nil);
    }

But at that screen I get printerManager nil

KevinGong2013, 2026-10-17:
Compatibility is tracked in a single pinned issue from now on; 5.0.0 scans for the common BLE printer services (Gprinter/Xprinter `E7810A71…`, ISSC `49535343…`, `18F0`, `FFE0`, `FF00`) and picks the write characteristic by its properties, so most ESC/POS printers with a BLE module work out of the box. Printers that only offer classic Bluetooth (SPP) cannot be used from iOS. Please add your model and result to the compatibility issue.

closed 2026-10-17 by KevinGong2013