KevinGong2013/Printer #63: How to print QRCode with Chinese content

caichunjian520, 2022-11-25:
All works fine with English Characters. If I change the text to Chinese, the QRCode won't print

KevinGong2013, 2023-02-03:
```
let receipt = Receipt(.🖨️58(String.GBEncoding.GB_18030_2000))
<<< QRCode(content: "你好，世界")
<<~ .page(.printAndFeed(lines: 0))
        
let data = Data(receipt.data)
dummyPrinter.write(data)
```

`Printer` framework has been rewrite.

PS: `String.GBEncoding.GB_18030_2000` is key point.

KevinGong2013, 2026-10-17:
5.0.0 fixes the QR command sequence (the model / module-size / print function codes were wrong, so many printers printed nothing) and adds a `Barcode` item (CODE128, EAN-13/8, UPC, CODE39, ITF, CODABAR, CODE93). For non-ASCII QR content use `QRCode(content: "中文", encoding: .utf8)`. Closing; reopen if it still fails on your printer.

closed 2026-10-17 by KevinGong2013