KevinGong2013/Printer #66: How to Print Arabic Language

manojssas, 2023-02-05:
How can I print Arabic Language with Combination of English and Image?

KevinGong2013, 2023-02-06:
`let receipt = Receipt(.🖨️58(.windowsCP1254))`

manojssas, 2023-02-06:
@KevinGong2013  When we using this. The Print comment Goes to blank [] block to print Like

extension String: ReceiptItem {
    
    public func assemblePrintableData(_ profile: PrinterProfile) -> [UInt8] {
        guard let data = self.data(using: profile.encoding) else {
            return []
        }
        return Array<UInt8>(data) + Command.CursorPosition.lineFeed.value
    }
}

I am using this code :

        let receipt = Receipt(.🖨️58(String.ArabicEncoding.windowsArabic))
                <<< "٥ حلواني"
                <<< KV("٢ برجر", "٢ برجر")

        if bluetoothPrinterManager.canPrint {
            bluetoothPrinterManager.write(Data(receipt.data))
        }

public extension String {
    
    struct ArabicEncoding {
        public static let windowsArabic = String.Encoding(rawValue: CFStringConvertEncodingToNSStringEncoding(CFStringEncoding(CFStringEncodings.windowsArabic.rawValue)))
    }
}

But this is not working
[image]
[image]
[image]

manojssas, 2023-02-08:
@KevinGong2013 Can you check and update me?
Thanks

KevinGong2013, 2023-02-08:
You should confirm your device support `Arabic`.

manojssas, 2023-02-08:
@KevinGong2013  Yes my device and printer both supports Arabic.
App was live from previous 5 Years on App store which has support for Arabic language.
And if we talk about the Printer. I am using X-Printer which supports Arabic language because in Android we are able to print Arabic language. Only iOS has issues.

manojssas, 2023-02-11:
@KevinGong2013 Hi is there any way to Print Arabic language. 
Thanks in Advance

manojssas, 2023-03-10:
Dear @KevinGong2013 
Have you got any idea like how to print Arabic language?
I do many changes inside Project but in Arabic mode this will print only blank.

Thanks

KevinGong2013, 2026-10-17:
Fixed in 5.0.0. The `ESC t` code-page byte was wrong before, so selecting a code page never worked. See the "Encodings" table in the README: pair the receipt encoding with the printer code page, e.g. `Receipt(.🖨️58(.windowsCP1252))` + `.style(.selectCharacterCodePage(.wpc1252))` for ä ö ü €, or `String.ArabicEncoding.windowsArabic` + `.wpc1256` for Arabic. Characters the encoding cannot represent are now replaced instead of dropping the whole line. Cash drawer: `.peripheral(.cashDrawer())`.

closed 2026-10-17 by KevinGong2013