KevinGong2013/Printer #67: Not able to print Image and Qr code

saniaaaaa, 2023-03-08:
Hi,
Not able to print Image and Qr code, it's working fine for text

i am using Printer Panda PRJ-R80B,

here's the code
let receipt = Receipt(.init(maxWidthDensity: 500, fontDesity: 12, encoding: .utf8))
<<~ .style(.initialize)
 <<< Image(cgImage)
<<< QRCode(content: "https://www.yuxiaor.com")
<<~ .page(.printAndFeed(lines: 2))

if bluetoothPrinterManager!.canPrint {
   bluetoothPrinterManager!.write(Data(receipt.data))
}
                    
dummyPrinter!.write(Data(receipt.data))

and this is the output I get
[image]

Please help me,
Thank you

Muhammadbarznji, 2023-08-15:
About. image issue rename it, because the Image is swiftUI struct.

KevinGong2013, 2026-10-17:
Fixed in 5.0.0 — the image path was rewritten: correct gray conversion (the old formula capped gray at 106, so the default threshold produced a blank image), any `CGImage` format (no more "Index out of range"), auto-scale to the paper width, banded `GS v 0` output, Floyd–Steinberg dithering, and Bluetooth writes are now split into 20-byte packets (large packets are silently truncated by many printer modules — verified on a Gprinter GP-58). QR codes use the correct command sequence as well.

Please update to 5.0.0 (Swift Package Manager; CocoaPods was dropped) and reopen if it still fails — include the base64 from `DummyPrinter().history` and your printer model.

5.0.0 fixes the QR command sequence (the model / module-size / print function codes were wrong, so many printers printed nothing) and adds a `Barcode` item (CODE128, EAN-13/8, UPC, CODE39, ITF, CODABAR, CODE93). For non-ASCII QR content use `QRCode(content: "中文", encoding: .utf8)`. Closing; reopen if it still fails on your printer.

closed 2026-10-17 by KevinGong2013