KevinGong2013/Printer #68: Unable to print long image

aprior199, 2023-04-07:
I need to print a full receipt like a long picture.

And printing stops at about 10%

Also I tried to cut picture for pieces, it helps but I have whitespaces between pictures

`let images = slice(image: resizedImage, into: 3)
        
        var ticket = Ticket()
        
        images.forEach {
            ticket.add(block: .image($0, attributes: .alignment(.center)))
        }
        
        ticket.feedLinesOnHead = 0
        ticket.feedLinesOnTail = 0
        
        if bluetoothPrinterManager.canPrint {
            bluetoothPrinterManager.print(ticket)
        }`

@KevinGong2013

burin-t, 2023-04-12:
haha just like me 
Work Around: separate image example 100px per image and add in ticket

KevinGong2013, 2026-10-17:
Fixed in 5.0.0 — the image path was rewritten: correct gray conversion (the old formula capped gray at 106, so the default threshold produced a blank image), any `CGImage` format (no more "Index out of range"), auto-scale to the paper width, banded `GS v 0` output, Floyd–Steinberg dithering, and Bluetooth writes are now split into 20-byte packets (large packets are silently truncated by many printer modules — verified on a Gprinter GP-58). QR codes use the correct command sequence as well.

Please update to 5.0.0 (Swift Package Manager; CocoaPods was dropped) and reopen if it still fails — include the base64 from `DummyPrinter().history` and your printer model.

closed 2026-10-17 by KevinGong2013