KevinGong2013/Printer #70: How we can connect wifi printers?

SHEBINKUNIYIL, 2023-05-16:
How we can connect wifi printers?

YanVetrov, 2023-05-19:
Example of direct tcp connection
```
import Network
import MobileCoreServices

                let PORT = NWEndpoint.Port(printerPort)
                let ipAddress = NWEndpoint.Host(printerIP)
                let queue = DispatchQueue(label: "TCP Client Queue")

                let tcp = NWProtocolTCP.Options.init()
                tcp.noDelay = true
                let params = NWParameters.init(tls: nil, tcp: tcp)
        connection = NWConnection(to: NWEndpoint.hostPort(host: ipAddress, port: PORT ?? 9100), using: params)
        connection?.stateUpdateHandler = { (newState) in

                    switch (newState) {
                    case .ready:
                        print("Socket State: Ready")
                        UserDefaults.standard.set(true, forKey: "isConnected")
                    default:
                        UserDefaults.standard.set(false, forKey: "isConnected")

                        break
                    }
                }
        connection?.start(queue: queue)

    func sendMSG(_ content:Data? = nil) {
           print("send data")
        connection?.send(content: content, completion: NWConnection.SendCompletion.contentProcessed(({ (NWError) in
               if (NWError == nil) {
                   print("Data was sent to TCP destination ")
                   
               } else {
                   print("ERROR! Error when data (Type: Data) sending. NWError: \n \(NWError!)")
               }
           })))
       }
```

SHEBINKUNIYIL, 2023-05-25:
thank you for the update, 
but Arabic text and image,s not printing

KevinGong2013, 2026-10-17:
Added in 5.0.0: `NetworkPrinter(host: "192.168.1.50")` sends ESC/POS over TCP port 9100, which Ethernet / Wi-Fi receipt printers listen on. On macOS there is also `SerialPrinter` for USB-serial printers.

closed 2026-10-17 by KevinGong2013